RobotecAI/rai #172: Create ros2 launch file to run rosbot XL demo

boczekbartek, 2024-08-29:
**Is your feature request related to a problem? Please describe.**
Currently, rosbot-XL demo requires running the simulation binary, navigation stack, and RAI node separately.

**Describe the solution you'd like**
It would be convenient to run this demo using the `ros2 launch` command.

**Describe alternatives you've considered**
- 

**Additional context**
Current workflow is described in: https://github.com/RobotecAI/rai/blob/development/docs/demos.md

iampaulidrobo, 2024-10-15:
Hello,
Can I work on this?

maciejmajek, 2024-10-16:
Hi @iampaulidrobo. Thank you for taking interest in RAI. This launch file is crucial for our ROSCon2024 demos and therefore will be assigned to @boczekbartek. Is there anything else you would like to tackle?

boczekbartek, 2024-11-12:
This issue has been resolved by: https://github.com/RobotecAI/rai/pull/291

closed 2024-11-12 by boczekbartek