RobotecAI/rai #204: Add HRI use as an optional (but recommended) way to run ROSBot example

adamdbrw, 2024-09-05:
**Is your feature request related to a problem? Please describe.**
Our key example is not using one of important features of RAI, which is Human-Robot Interface.

**Describe the solution you'd like**
Update the example, developing HRI support. This could be text to start with, but it would be best to be able to try out voice interaction with the simulated demo using your device mic & speaker.

**Describe alternatives you've considered**
Developing a separate HRI demo, but this would be artificial.

**Additional context**
It makes sense for some applications to use smartphone to communicate with robots.

maciejmajek, 2025-02-07:
Some effort has been started under #411. The next step is to isolate the dependencies

maciejmajek, 2025-04-17:
Outdated

closed 2025-04-17 by maciejmajek