RobotecAI/rai #272: Store information about vector database embedding model

maciejmajek, 2024-10-11:
**Is your feature request related to a problem? Please describe.**
Currently, the whoami packages are created using openai's embedding models. Using those with other vendors is impossible.

**Describe the solution you'd like**
Internal tools used for rag should make sure that the model is compatible with the vdatabase.  

**Describe alternatives you've considered**
Creating a log file with information about the generation (vendor, models etc)

**Additional context**
Add any other context or screenshots about the feature request here.

boczekbartek, 2024-12-11:
@maciejmajek 
I think current `development` branch allows using different embedding model than `openai`. See:
- [config.toml](https://github.com/RobotecAI/rai/blob/d15910ad4af54d7574a96fac3dbf4eceacd8b8e2/config.toml#L4)
- Turtlebot demo already uses local embeddings in the example (point 3 [here](https://github.com/RobotecAI/rai/tree/development/src/examples/turtlebot4#step-by-step-instructions))

maciejmajek, 2024-12-11:
Yup, user can configure the vendor now. However if the config (embeddings model) would change after the vdb is created, the queries won't work, as the embeddings in the database were created using different models.