RobotecAI/rai #292: Having troubles with the demos? Please read

maciejmajek, 2024-10-22:
# ROSCon 2024 Demos
We have released several demos that were showcased at ROSCon 2024. These demos have been  tested internally on the following configurations:
- Ubuntu 22.04 with ROS 2 Humble
- Ubuntu 24.04 with ROS 2 Jazzy

**If you encounter any issues or bugs during the setup or execution of the demos, please don't hesitate to:**
- Report them in this thread, or
- Create a new Issue in the repository
Your feedback is greatly valued!

boczekbartek, 2024-11-28:
@maciejmajek 
I was testing agriculture demo while working on https://github.com/RobotecAI/rai/pull/325, but I got an error connected with `rai_interfaces` error. Do you know what might be the reason?

```
2024-11-28 10:20:44 robo-pc-005 rai.utils.model_initialization[330636] INFO Using LLM model: openai-gpt-4o-mini
[INFO] [1732785648.080175529] [mock_behavior_tree_node_1]: Current state: DRIVING: the vehicle is performing a mission
2024-11-28 10:20:50 robo-pc-005 httpx[330636] INFO HTTP Request: POST https://api.openai.com/v1/chat/completions "HTTP/1.1 200 OK"
[INFO] [1732785650.794136301] [rai_node]: Topic '/tractor1/camera_image_color' postprocessed in: 5.81
2024-11-28 10:20:51 robo-pc-005 httpx[330636] INFO HTTP Request: POST https://api.openai.com/v1/chat/completions "HTTP/1.1 200 OK"
[INFO] [1732785651.810960019] [rai_node]: Logs summary retrieved in: 1.02
2024-11-28 10:20:51 robo-pc-005 rai.utils.model_initialization[330636] INFO Using LLM model: openai-gpt-4o-mini
[INFO] [1732785653.080285975] [mock_behavior_tree_node_1]: Current state: STOPPED: the vehicle has stopped
[INFO] [1732785653.081014320] [mock_behavior_tree_node_1]: The tractor has stopped. Calling RaiNode to decide what to do.
python: /home/bboczek/projects_u24/01_internal/01_repos/12_rai/build/rai_interfaces/rosidl_generator_py/rai_interfaces/action/_task_s.c:91: rai_interfaces__action__task__goal__convert_from_py: Assertion `PyUnicode_Check(field)' failed.
```

boczekbartek, 2024-11-28:
@maciejmajek I think the reason of the error is that in [rai_interfaces.action.Task](https://github.com/RobotecAI/rai/blob/development/src/rai_interfaces/action/Task.action) `priority` is a string, but in agri demo, it's set to be an int. I'll create a PR with a fix soon

maciejmajek, 2025-04-17:
Outdated, will be moved to discussion board.

closed 2025-04-17 by maciejmajek

boczekbartek, 2025-04-17:
Feel free to join discord server and ask questions there: https://discord.gg/3PGHgTaJSB