RobotecAI/rai #313: improve documentation of rai tool-calling

boczekbartek, 2024-11-22:
**Is your feature request related to a problem? Please describe.**
`rai` currently has multiple levels of tools.

1. Generic tools that call ros2 topics/services/actions 
2. Specialized tools (like nav2 tools or open-set detection tools) that implement their own node
3. Different types of tools attached to RaiNode are handled differently due to dedicated node spinning (`rclpy.spin`) implementation and to avoid deadlock. Tools that connect to topics are handled using different node to tools that call services 

**Describe the solution you'd like**
Such dedicated implementations should be documented so that developer knows on which level a new tool should be implemented.

**Describe alternatives you've considered**
Implement general solution so that tool type choice is not important from the higher level developer-facing API. This solution would be better but requires more time.

**Additional context**
Add any other context or screenshots about the feature request here.

rachwalk, 2025-01-28:
I think this issue has been fully addressed by https://github.com/RobotecAI/rai/pull/383 and addition of connectors. If I'm correct, can you close it? @boczekbartek

boczekbartek, 2025-01-28:
Yes, I think we can resolve this issue, but I think it's more #335.
In #383 the work of 335 is continued/majorly refactored and integrated with connectors, but is not yet integrated with LLM agents and demo projects.

closed 2025-01-28 by boczekbartek