RobotecAI/rai #345: integrate with LeRobot

boczekbartek, 2024-12-11:
**Is your feature request related to a problem? Please describe.**
Currently [rai-manipulation-demo](https://github.com/RobotecAI/rai/blob/development/docs/demos/manipulation.md) uses ros2 moveit and a fusion of segmentation with depth camera to control Panda robotic arm. 
This approach is good to for reasoning and planning, but might be limited when it comes to more complex movements of the gripper. For such task, policies implemented in [LeRobot](https://github.com/huggingface/lerobot) can be more useful.

LeRobot currently offers:
- 4 pre-trained policies listed in Models section: [link](https://huggingface.co/lerobot)
- scripts for policies training: [link](https://github.com/huggingface/lerobot?tab=readme-ov-file#train-your-own-policy)
- dataset format for training policies: [link](https://github.com/huggingface/lerobot?tab=readme-ov-file#the-lerobotdataset-format)

**Describe the solution you'd like**
I find the integration of all LeRobot features mentioned above useful in `rai`, but I think a step-by-step should be applied.
- First logical step would be to attach pre-trained policy to the rai agent as a tool.
- Next would be to determine what policies would be useful for manipulation agent and to train them on datasets curated in simulation.
- Next step would be to record generate trajectories for training using rai and text descriptions

**Describe alternatives you've considered**

**Additional context**

vmayoral, 2024-12-13:
Hi there,

I'm happy spending some cycles in exploring and potentially contributing to this. I've got some experience with LeRobot (participated in [their last hackathon last Oct](https://github.com/huggingface/lerobot_hackathon_oct2024) and ported it to a new robot as part of the effort which I [contributed back](https://github.com/huggingface/lerobot/pull/493)) which should help in this endeavour.

> - First logical step would be to attach pre-trained policy to the rai agent as a tool.
> - Next would be to determine what policies would be useful for manipulation agent and to train them on datasets [...]

For now, I'll take on the first task, which is in-line with my research interests. I can't commit to the other two now but I'm happy to see how far my cycles can take me. Question though, what do you mean by:

`Next step would be to record generate trajectories for training using rai and text descriptions`

Don't quite get the meaning behind it.

I may have some questions @boczekbartek concerning the [rai-manipulation-demo](https://github.com/RobotecAI/rai/blob/development/docs/demos/manipulation.md). I guess the discord server of the Embodied AI WG is the best place for discussion?

boczekbartek, 2024-12-13:
@vmayoral 
That's awesome to hear! Having someone with hands on experience with LeRobot to work on this issue is a best case scenario. Your contribution here would be very welcome! 

> Next step would be to record generate trajectories for training using rai and text descriptions

( _I adjusted the wording a bit in the description to make it more clear._)

My idea for this was to use the pipeline of the current [rai-manipulation-demo](https://github.com/RobotecAI/rai/blob/development/docs/demos/manipulation.md) (with LLM tools that are based on moveit and groundedSAM) to generate trajectories for imitation learning in LeRobot. From training on these trajectories we could get more robust policies. 
Does it make sense to you? 

Also, if you have suggestions for the trajectory for this issue please share them. We can figure out together what makes the most sense.

As for the [rai-manipulation-demo](https://github.com/RobotecAI/rai/blob/development/docs/demos/manipulation.md) (and basically other questions about rai / llm agents ) - yes, please write us on Discord. For the demo you could also use this issue: https://github.com/RobotecAI/rai/issues/292. We'll be happy to help!

vmayoral, 2024-12-18:
>  Does it make sense to you?

Thanks, it does now.

Looking to engage into all of this later this week. Will report back here.

vmayoral, 2025-01-18:
Hi all,

Thought it took a bit longer than I was expecting initially, I was positively hoping to report a small success integrating a LeRobot policy into RAI. Unfortunately this turned frustrating Altogether, **I have not been able to integrate LeRobot policies into RAI** ❌ because honestly, I have not been able to get RAI to operate. <ins>No blaming, just reporting my personal experience here and hoping to retry again in the future if time allows for it</ins>. Here's my experience:

#### Setting up the policy and tools for "integration"

I built a small policy using `ACT` (one of the provided algorithms within `LeRobot`) that successfully converges both gradient and loss-wise. The performance of the policy is reasonable and "prepared" to be chain with other traditional manipulation behaviors leveraging _optimal control_ (e.g. via ROS 2). I've worked with real robots, as it's more fun. Here's the current policy operating to place the rubber plate in the desired position. Handling such material is rather challenging generally:

| top  | left | right |
|-----|-----|------|
| https://github.com/user-attachments/assets/cdb87412-435d-4754-b7ee-ace9ee7b6f45 | https://github.com/user-attachments/assets/feb2c99a-76bd-4e61-a10f-94275ce7ddc3 | https://github.com/user-attachments/assets/b3f5b31e-8717-4808-9a80-88e1a2e47924 |

With this operational, I tested this outside of RAI and ROS 2. A simple Python script can trigger this policy by leveraging the [`control_loop`](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/control_utils.py#L228-L293) function `LeRobot` provides while passing the right trained policy (or if you wish to leverage  higher level control, you can look into `LeRobot`'s [`record`](https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/control_robot.py#L187-L333))

#### Integrating the policy into `RAI`

Unfortunately, things turned difficult here. The struggles with RAI got me to spend a very significant amount of time on troubleshooting. I'm listing below my approach and some notes of the process:

- Followed https://github.com/RobotecAI/rai?tab=readme-ov-file#setup
- Then followed https://github.com/RobotecAI/rai/blob/development/docs/demos/manipulation.md#setup

Major issues experienced:

<details><summary>:x: Issue 0: a) while launching configurator.py, PortAudio missing and b) later further issues</summary>

```bash
2025-01-18 10:46:50.453 Uncaught app execution
Traceback (most recent call last):
  File "/home/dragon/.cache/pypoetry/virtualenvs/rai-3Vgzn0Ug-py3.12/lib/python3.12/site-packages/streamlit/runtime/scriptrunner/exec_code.py", line 88, in exec_func_with_error_handling
    result = func()
             ^^^^^^
  File "/home/dragon/.cache/pypoetry/virtualenvs/rai-3Vgzn0Ug-py3.12/lib/python3.12/site-packages/streamlit/runtime/scriptrunner/script_runner.py", line 579, in code_to_exec
    exec(code, module.__dict__)
  File "/home/dragon/rai/src/rai/rai/utils/configurator.py", line 21, in <module>
    import sounddevice as sd
  File "/home/dragon/.cache/pypoetry/virtualenvs/rai-3Vgzn0Ug-py3.12/lib/python3.12/site-packages/sounddevice.py", line 71, in <module>
    raise OSError('PortAudio library not found')
OSError: PortAudio library not found
^C  Stopping...
```

```bash
poetry run streamlit run src/rai/rai/utils/configurator.py
Traceback (most recent call last):
  File "/home/dragon/.local/share/pypoetry/venv/lib/python3.12/site-packages/cleo/application.py", line 327, in run
    exit_code = self._run(io)
                ^^^^^^^^^^^^^
  File "/home/dragon/.local/share/pypoetry/venv/lib/python3.12/site-packages/poetry/console/application.py", line 188, in _run
    self._load_plugins(io)
  File "/home/dragon/.local/share/pypoetry/venv/lib/python3.12/site-packages/poetry/console/application.py", line 354, in _load_plugins
    manager.load_plugins()
  File "/home/dragon/.local/share/pypoetry/venv/lib/python3.12/site-packages/poetry/plugins/plugin_manager.py", line 38, in load_plugins
    self._load_plugin_entry_point(ep)
  File "/home/dragon/.local/share/pypoetry/venv/lib/python3.12/site-packages/poetry/plugins/plugin_manager.py", line 76, in _load_plugin_entry_point
    plugin = ep.load()  # type: ignore[no-untyped-call]
             ^^^^^^^^^
  File "/home/dragon/miniconda3/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dragon/miniconda3/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/dragon/.local/share/pypoetry/venv/lib/python3.12/site-packages/poetry_plugin_export/plugins.py", line 7, in <module>
    from poetry_plugin_export.command import ExportCommand
[... 333 lines cut ...]
    from rai.agents.conversational_agent import create_conversational_agent
[... cut ...]
At this point I run out of time (and patience). It **took me much (much!) longer the troubleshooting of setting up RAI**  than I spent with the rest of the tasks (meaning setting up the robots, recording episodes through Imitation Learning, training, testing the policy and creating a silly wrapper for launching such policy at desire from a python wrapper with 10 LOC).

That said, I deeply respect and appreciate the tremendous effort and ingenuity that goes into creating tools like RAI. They are foundational to the field and enable so much innovation.

maciejmajek, 2025-01-18:
Hi @vmayoral,

Thank you for sharing your detailed experience. I understand that you might have had Conda sourced or used it instead of Poetry. Unfortunately, we don’t support Conda and haven’t tested RAI in that environment. The framework is officially supported and tested with Poetry, and the issues you’ve described are likely linked to incompatibilities introduced by having Conda sourced during the setup.

The only issue I can directly relate to is the empy error, which can occur when building the repository for the first time in a custom Python environment. This is usually caused by package overshadowing, and we resolve it as outlined here: RAI Discussion #150.

We’ve also received feedback from the community regarding challenges in using RAI, but, the problems were not related to the Python environment itself but to other aspects of the process. However, having Conda sourced can introduce subtle conflicts, so I’d like to clarify: Did you have a Conda environment active during your setup, even if you didn’t explicitly use Conda for the installation?

If Conda is your preferred approach, we’d be open to orchestrating support for it in some capacity, as we understand the importance of adapting to diverse workflows. Your feedback is invaluable, and we truly appreciate the effort you’ve put into documenting your experience.

Thank you for your patience and for the kind words about the work behind RAI.

vmayoral, 2025-01-19:
@maciejmajek thanks for the prompt response!

As you hint, I indeed had a `conda` env. Reason behind this is that despite `pip install -e .` gets the source code installed, gettings things to work an in-line with upstream development is a completely different story. For what's worth, throughout my troubleshooting sessions, made various attempts to remove conda. I also tried a different approach which was to seek python compatibility via a standalone version of Python (through `pyenv`, which I was hoping could marry better with poetry). 

To my understanding, the whole issue boils down to a python compatibility issue, which I have not been able to solve.

> To ensure that the installation process works as expected, we have continuous integration (CI) workflows in place. These CI pipelines are based on clean humble and jazzy OSRF Docker images. They test the entire installation process, verify that all packages can be imported correctly, and run a portion of RAI to confirm its functionality. This ensures that the framework is operable in the intended environment.

Is it possible to disclose such Dockerfile (or a simplified version of it) that helps others (e.g. me) reproduce the process of getting RAI up to speed?

maciejmajek, 2025-01-19:
> As you hint, I indeed had a conda env. Reason behind this is that despite pip install -e . gets the source code installed, gettings things to work and in-line with upstream development is a completely different story, at least in my experience with LeRobot.

I see... 

> Is it possible to disclose such Dockerfile (or a simplified version of it) that helps others (e.g. me) reproduce the process of getting RAI up to speed?

Yes, I will provide the Dockerfile first thing tomorrow and possibly host the built image later on.

maciejmajek, 2025-01-19:
Effort started: https://github.com/RobotecAI/rai/pull/377
I think we may also push the images to ghcr/docker registry. Will let you know.

maciejmajek, 2025-01-19:
Using the Humble Docker image from #377, I was able to execute the following:

```bash
# /rai
poetry shell
git clone lerobot
cd lerobot
pip install -e .
pytest # Run lerobot tests
cd ../
pytest tests # Run rai tests
```

Although the `pip install -e .` command caused some package conflicts, the tests for `lerobot` succeeded, while the `rai` tests failed. However, the current error in the `rai` tests should not cause any issues with using the framework.

Test results for `lerobot`:
```bash
============================== 161 passed, 150 skipped, 5 warnings in 165.17s (0:02:45) ==============================
```

Test results for `rai`:
```bash
=============== 1 failed, 29 passed, 1 skipped, 6 deselected, 1 xpassed, 2 warnings, 1 error in 13.65s ===============
```

maciejmajek, 2025-01-21:
Hi @vmayoral, we have made some progress with docker setup. We tested it a bit and it seems promising. https://github.com/RobotecAI/rai/issues/376

Also, fyi @boczekbartek tried running RAI solely on Conda, but due to conda's nature it turned out to be impossible :(

boczekbartek, 2025-01-21:
That's true, I tried Conda setup and failed. The problem seem to lay in the `conda`~`ros2` conflict. Here's what I did:
1. Created fresh conda env: `conda create -y -n my_env python=3.12`
2. Installed `lerobot` with `pip install -e`
3. With some modifications to `pyproject.toml` I managed to install rai python dependencies (not fully finished yet, I need to work on it more, but basically adaptation to [PEP621](https://peps.python.org/pep-0621/) )
4. managed to build rai ros2 packages with colcon. It's important to do `conda deactivate` before `colcon build --symlink-install`, to avoid the `em` error.
5. Failed to run any python code that requires `rclpy`. It fails on `import rclpy` , because of some libraries conflict. I consistently get this error:

  <details>

  ```
  Python 3.12.8 | packaged by Anaconda, Inc. | (main, Dec 11 2024, 16:31:09) [GCC 11.2.0] on linux
  Type "help", "copyright", "credits" or "license" for more information.
  >>> import rclpy
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 49, in <module>
      from rclpy.signals import install_signal_handlers
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/signals.py", line 15, in <module>
      from rclpy.exceptions import InvalidHandle
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/exceptions.py", line 16, in <module>
      from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
      rclpy_implementation = import_c_library('._rclpy_pybind11', package)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
      return importlib.import_module(name, package=package)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/bboczek/anaconda3/envs/lerobot4/lib/python3.12/importlib/__init__.py", line 90, in import_module
      return _bootstrap._gcd_import(name[level:], package, level)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  ImportError: /home/bboczek/anaconda3/envs/lerobot4/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /opt/ros/jazzy/lib/python3.12/site-packages/rclpy/_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so)
  The C extension '/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-even-with-library-present-on-the-system' for possible solutions
  ```

  </details> 

For point 5:
 * I tried different orders of sourcing: conda and ros2 (`conda activate my_env`, `. ./install/setup.bash`)
 *  even tried different order of paths (`conda_path:ros2_paths` or `ros2_paths:conda_paths`) in `$PATH` and `$LD_LIBRARY_PATH`  

I'm afraid to integrate `lerobot` with `rai`, conda is not a good fit, because of ros2. I would recommend getting inspired by docs and dockerfile, installing rai locally with poetry and following what @maciejmajek posted in [his comment](https://github.com/RobotecAI/rai/issues/345#issuecomment-2601044297) .

vmayoral, 2025-02-09:
Thanks a lot @maciejmajek and @boczekbartek for helping out here. Your efforts are rather welcome. Let me address some of the progress and support:

@boczekbartek thanks for the attempts at merging/mixing things. Your report above resonates with my previous attempt [here](https://github.com/RobotecAI/rai/issues/345#issuecomment-2599686208), leading (from a quick look) to similar roadblocks. I agree with your conclusion, the holistic/dockerized approach is probably a much better approach so I've pivoted to that now that you guys have provided some pointers there.

@maciejmajek, went ahead and tried out the Docker image (thanks again for that, helpful!). Image buils fine and then followed https://github.com/RobotecAI/rai/blob/development/docs/setup_docker.md#3-run-the-tests-to-confirm-the-setup to test the setup. That led me to into a direct issue ⚠ :

[image]

AFAIK, issue there is a common Docker development challenge where you need to:
- Use the source code inside the Dockerfile to build dependencies.
- Still mount the source code during runtime for easy development without overwriting built artifacts.

This can be tackled with a two-staged build approach, or simple by invoking the `docker run` without the volume mount:
```bash
docker run --net=host --ipc=host --pid=host -it rai:humble
# as opposed to 
#   docker run --net=host --ipc=host --pid=host -it -v $(pwd):/rai rai:humble
```

Altogether:

[image]

Send a small contribution at https://github.com/RobotecAI/rai/pull/413

vmayoral, 2025-02-09:
@maciejmajek 

> ```bash
> # /rai
> poetry shell
> git clone lerobot
> cd lerobot
> pip install -e .
> pytest # Run lerobot tests
> cd ../
> pytest tests # Run rai tests
> ```
> 

This did not work for me. In particular, it directly fails the `rai` tests:

```bash
dragon@dragon:~/rai$ docker run --net=host --ipc=host --pid=host -it rai:humble
root@dragon:/rai# poetry shell
Spawning shell within /root/.cache/pypoetry/virtualenvs/rai-LG7o02pr-py3.10
root@dragon:/rai# . /root/.cache/pypoetry/virtualenvs/rai-LG7o02pr-py3.10/bin/activate
(rai-py3.10) root@dragon:/rai# pytest
ImportError while loading conftest '/rai/tests/conftest.py'.
tests/conftest.py:26: in <module>
    from rai.config.models import BEDROCK_CLAUDE_HAIKU, OPENAI_MINI
E   ModuleNotFoundError: No module named 'rai'
```

maciejmajek, 2025-02-09:
@vmayoral just a quick reply, more will follow during the week. This happens when source setup shell has not been invoked previously. First step is to build outside of the sourced shell and then run the source script. Will take a look at it again tomorrow.

vmayoral, 2025-02-09:
> [@vmayoral](https://github.com/vmayoral) just a quick reply, more will follow during the week. This happens when source setup shell has not been invoked previously. First step is to build outside of the sourced shell and then run the source script. Will take a look at it again tomorrow.

Thanks, that does it:

```bash
dragon@dragon:~/rai$ docker run --net=host --ipc=host --pid=host -it rai:humble
root@dragon:/rai# source setup_shell.sh 
(rai-py3.10) root@dragon:/rai# poetry shell
Spawning shell within /root/.cache/pypoetry/virtualenvs/rai-LG7o02pr-py3.10
root@dragon:/rai# . /root/.cache/pypoetry/virtualenvs/rai-LG7o02pr-py3.10/bin/activate
(rai-py3.10) root@dragon:/rai# pytest tests
...
(tests OK)
```