RobotecAI/rai #381: RFC: ROS2ARIConnector callback behavior

maciejmajek, 2025-01-21:
### Description
Currently, the callback functions passed to start_action are executed using `ThreadExecutor`. The following test checks whether a callback can prevent the program from exiting:  
https://github.com/RobotecAI/rai/blob/c6c49b967d6df8ba7b26decbbd92c902b58a4d33/tests/communication/ros2/test_connectors.py#L168-L194
(This test is skipped as indeed, this stops pytest from exiting before the 100s mark)

However, it is challenging to ensure that the callback is implemented correctly so that killing the thread leaves no zombie processeses remaining.  

Given this complexity, I propose delegating the responsibility of writing a proper callback function to the user and removing this essentially useless test.

rachwalk, 2025-01-28:
In general there is no way to ensure that a callback terminates and performs a clean exit. The drawback of callbacks is that the API has no control over them (as opposed to say threads, which can be explicietly `join()`ed by the API), but that comes with more flexibility to API users. Because of that having a callback based design puts the responsibility of writing correct callbacks on the user. Consider the following case (pseudocode):

```
thread_registry = [Thread(target=some_func1), Thread(target=some_func2), Thread(target=some_func3)]

def callback(idx: int):
     thread_registry[idx].start()

handle = connector.start_action(data, name, on_feedback=callback)
# some waiting
connector.stop_action(handle)
for thread in thread_registry:
    thread.join()
``` 

This is a perfectyly valid use case (starting different threaded functions on action feedback), and what is more the user performs correct cleanup of the threads post action execution. Note that within the scope of the connector there is no access or knowledge to the threads, thread_registry or the cleanup step. It is also in my opinion unreasonable to anticipate that the on_feedback callback of the action terminates before the action itself - the handling of feedback can finish much later than the action.

Also please note, that the test is not aptly named - the callback call is not `blocking` it merely terminates after the action is finished.

To address your concern instead of callbacks we would have to explicitely use a mechanism like threads or async/await which can be explicitely terminated, but that would severely limit API functionality, while not providing (in my opinion) a lot of value. For the abovementioned reasons I believe that this test should be removed and the issue closed. I have created a PR regarding the isssue: https://github.com/RobotecAI/rai/pull/392

maciejmajek, 2025-01-28:
Agreed

rachwalk, 2025-01-28:
Closing since https://github.com/RobotecAI/rai/pull/392 was merged.

closed 2025-01-28 by rachwalk