RobotecAI/rai #405: RAI Benchmark: Architecture proposal

maciejmajek, 2025-02-06:


MagdalenaKotynia, 2025-02-06:
Current draft proposal: https://github.com/RobotecAI/rai/blob/feat/benchmarks/src/rai_benchmarks/benchmark_model.py

MagdalenaKotynia, 2025-02-07:
Redefined [draft of benchmark design](https://github.com/RobotecAI/rai/blob/feat/benchmarks/src/rai_benchmarks/benchmark_model.py). 

It is based on the following assumptions:
* User defines:
     * agent
     *  tasks (prompt and appropriate metric. Metric can be based on the provided services e.g. service getting poses of the objects.), 
     * scene_configs defining e.g. start objects positions, 
     * scenarios (pairs of the tasks and scene_configs)
* Simulation needs to meet some assumptions defined by the specific EngineConnector - e.g. specific services (e.g. used for spawning objects getting objects names and poses, getting available objects to spawn); for O3DEngineConnector simulation provided as binary
* SceneConfig is a model that matches specific EngineConnector - e.g. when O3DEngineConnector has a possibility to spawn objects, O3DESceneConfig has field defining objects names and their initial poses.

boczekbartek, 2025-03-27:
@MagdalenaKotynia @maciejmajek Can we close this issue?

MagdalenaKotynia, 2025-03-27:
#436 addressing this issue has been merged to development, so I am closing the issue

closed 2025-03-27 by MagdalenaKotynia