RobotecAI/rai #408: Refactor ros2 tools used in demo manipulation demo to use `Ros2ARIConnector`

boczekbartek, 2025-02-06:
In [manipulation demo project](https://github.com/RobotecAI/rai/blob/development/docs/demos/manipulation.md) LLM agents can use ros2 tools. Ros2 communication inside these tools is currently based on the [RaiBaseNode](https://github.com/RobotecAI/rai/blob/6bae47f48f03fc6bad212766548ac82b90688f70/src/rai/rai/node.py#L118) that will be soon replaced by new APIs based on the the [base_connector.py](https://github.com/RobotecAI/rai/blob/development/src/rai/rai/communication/base_connector.py).

The aim of this issue is to:
- replace `RaiBaseNode` dependency from tools and implement them using [Ros2ARIConnector](https://github.com/RobotecAI/rai/blob/6bae47f48f03fc6bad212766548ac82b90688f70/src/rai/rai/communication/ros2/connectors.py#L39)
- if some tools are not used anymore in other demo projects - remove them

boczekbartek, 2025-04-17:
Most of the tools has been refactored in https://github.com/RobotecAI/rai/pull/521.
However one thing missing to resolve this issue is that tools from https://github.com/RobotecAI/rai/blob/28eb194a29320e4b8236813faea8ea52c249aede/src/rai_extensions/rai_open_set_vision/rai_open_set_vision/tools/segmentation_tools.py should inherit from `ROS2BaseTool`

maciejmajek, 2025-05-13:
This is done