RobotecAI/rai #464: Improve benchmark integration with langfuse/langsmith

MagdalenaKotynia, 2025-03-21:
**Is your feature request related to a problem? Please describe.**
Related to #455.
Currently implemented integration with langfuse and langsmith could be improved.

**Describe the solution you'd like**
Proposed improvements:
* Add error codes for errors categories (could be useful for e.g. easier quering of desired runs by their error/easier creating of datasets)
* Mark all runs coming from one benchmark run with session ID to easily get runs coming only from one benchmark run.

**Additional context**
Errors included in `Comment` column could have error category IDs. (screen from Langfuse)
[image]

Session column in langfuse could be used (in langsmith other mechanism could be used, I didn't notice explicit session id there)

[image]

maciejmajek, 2025-04-17:
@MagdalenaKotynia what's the current timeline for this task? Anyone working on this?

MagdalenaKotynia, 2025-04-18:
@maciejmajek I haven't started working on this task. This enhancement proposal was created as a future enhancement after more priority work on the tool calling benchmark is done. I suggest to start working on it after the refactor of rai_bench in #517 is finished and merged. 
I edited the issue and added some other proposed improvements related to tracing.

maciejmajek, 2025-05-03:
I'm removing this issue from RAI 2.0, due to time constraints as well as low priority of the task

jmatejcz, 2025-06-09:
applied here https://github.com/RobotecAI/rai/pull/606