RobotecAI/rai #490: Publish rai_interfaces to rosdep

maciejmajek, 2025-03-28:
**Is your feature request related to a problem? Please describe.**
As of now, to use full potential of RAI a repository has to be cloned and built via colcon.

**Describe the solution you'd like**
Users should be able to install packages that would otherwise had to be built using rosdep/apt.

```
sudo apt install ros-jazzy-rai-interfaces
```

The point of this issue is to find out the feasibility of publishing rai_interfaces to rosdep and optionally rai_openset

knicked, 2025-04-02:
I have put `rai_interfaces` into a [new repository](https://github.com/RobotecAI/rai_interfaces), and opened a [PR](https://github.com/ros/rosdistro/pull/44966) to rosdistro. Waiting for review...

maciejmajek, 2025-04-17:
@knicked any progress?

maciejmajek, 2025-05-13:
I can see, that the [latest pr](https://github.com/ros/rosdistro/pull/45711) has been merged. If all goes well, the package should be available in about 2 to 4 weeks.

closed 2025-05-13 by maciejmajek