RobotecAI/rai #497: GetROS2MessageInterfaceTool does not return types

jmatejcz, 2025-03-31:
**Is your feature request related to a problem? Please describe.**
The GetROS2MessageInterfaceTool return only fields with default values, omitting field types, which makes agent not able to deduce proper message structure.

for example message type rai_interfaces/srv/ManipulatorMoveTo when investigated via ros2cli outputs detailed info about every field:
[image]

but the tool outputs only fields names and defaults and when field is complex type like audios, the agent can't know how to fill it.
[image]

[image]

**Describe the solution you'd like**
Ideally the tool should output the types.

**Describe alternatives you've considered**

**Additional context**

maciejmajek, 2025-04-17:
@jmatejcz any updates?

jmatejcz, 2025-04-24:
added here:
https://github.com/RobotecAI/rai/issues/515

closed 2025-04-24 by jmatejcz