RobotecAI/rai #513: MultiThreadedExecutor in ROS2ARIConnector slow things down 100 times

maciejmajek, 2025-04-07:
**Describe the bug**
I was reviewing [this](https://github.com/RobotecAI/rai/pull/501) lovely PR which transforms our old openset detection and segmentation solution into agents.
I've noticed a critically bad performance and decided to dive into debugging.
Tried few things:
1. substitution wait_for_shutdown with time.sleep
2. Debugging weights loading
3. Finally decided to change the executor to SingleThreaded. The speed. The performance. The speed. The power. 

[image]

**To Reproduce**
Steps to reproduce the behavior:

1. git checkout [9cf286f](https://github.com/RobotecAI/rai/pull/501/commits/9cf286f2c5c99ded78a4c5f95b761449f1971d4f)
2. Run the following script 

```python
from rai_open_set_vision.agents import GroundingDinoAgent, GroundedSamAgent
from rai.utils import wait_for_shutdown
import rclpy

def main():
    rclpy.init()
    agent1 = GroundingDinoAgent()
    agent2 = GroundedSamAgent()
    agent1.run()
    agent2.run()
    wait_for_shutdown([agent1, agent2])
    rclpy.shutdown()

if __name__ == "__main__":
    main()
```
3. Run these tests
```python

from rai.communication.ros2 import ROS2ARIConnector, ROS2ARIMessage
from rai.utils import ROS2Context
from rai_interfaces.srv import RAIGroundingDino, RAIGroundedSam
import cv2
import cv_bridge

from rai.tools.ros2.utils import ros2_message_to_dict

@ROS2Context()
def test_grounding_dino_agent():
    connector = ROS2ARIConnector()
    bridge = cv_bridge.CvBridge()

    image = bridge.cv2_to_imgmsg(cv2.imread("docs/imgs/o3deSimulation.png"))

    msg = RAIGroundingDino.Request()
    msg.source_img = image
    msg.classes = ["chair"]
    msg.box_threshold = 0.5
    msg.text_threshold = 0.5
    
    ari_msg = ROS2ARIMessage(payload=ros2_message_to_dict(msg))
    response = connector.service_call(target="/grounding_dino_classify", message=ari_msg, msg_type="rai_interfaces/srv/RAIGroundingDino", timeout_sec=10)
    print(response.payload)

@ROS2Context()
def test_grounded_sam_agent():
    connector = ROS2ARIConnector()
    bridge = cv_bridge.CvBridge()

    image = bridge.cv2_to_imgmsg(cv2.imread("docs/imgs/o3deSimulation.png"), encoding="rgb8")

    msg = RAIGroundedSam.Request()
    msg.source_img = image
    msg.detections = {}

    ari_msg = ROS2ARIMessage(payload=ros2_message_to_dict(msg))
    response = connector.service_call(target="/grounded_sam_segment", message=ari_msg, msg_type="rai_interfaces/srv/RAIGroundedSam", timeout_sec=10)
    print(response.payload)
```
4. Note the time it took pytest to finish
5. Change the following line to SingleThreadedExecutor (remember to update import) https://github.com/RobotecAI/rai/blob/0675ff6036bbd1450f34426f10d64f999ab91777/src/rai_core/rai/communication/ros2/connectors/ari_connector.py#L106
6. Run the script and pytest again, note the time.
7. Should be 100 times faster.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform**

- OS: 24.04
- ROS 2 Jazzy

maciejmajek, 2025-04-07:
@rachwalk I think you've mentioned it working fairly well on ubuntu 22.04 with ROS 2 humble. Can you please run the pytest to check if changing the executor speeds things up?

adamdbrw, 2025-04-07:
Did you investigate why this is happening? Is there excessive synchronization? Does it boil down to executor internals?
There has been some fairly recent work on rclpy executors that could be an improvement once we get to Kilted: https://discourse.ros.org/t/faster-rclpy-executor-now-in-rolling/42852

maciejmajek, 2025-04-07:
I will delve into the executor in my free time. Unfortunately, the aforementioned executor treats everything as mutually exclusive, which is unsuitable for our connectors :(

maciejmajek, 2025-04-07:
I've tested the problem on a different Ubuntu 24.04 machine and the problem does not exist. We need to find a way to detect if such bottleneck exists as it directly impacts the swiftness of RAI.

rachwalk, 2025-04-09:
@maciejmajek Couldn't reproduce on Ubuntu 22.04. The two tests described in issue took on average 2.2s using `SingleThreadedExecutor` and 1.5s using the `MultiThreadedExecutor`. I found no difference between using the two executors, and couldn't reproduce this issue.